tronxy3d/F4xx-SIM480x320 #1: Cannot compile firmware for STM32F103 board

luigimannoni, 2022-11-13:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

I have a XY2-PRO with Titan extruder, compiling the firmware with the STM32F446 board as target does not update the printer, I believe I have a 103 board (2 capacitators - chitu v6)

If I change the `platformio.ini` file at row 17: `default_envs = tronxy_stm32f103`  the compilation fails with the following error:

```
Marlin\src\HAL\STM32\HAL.cpp: In function 'void HAL_SYSTICK_Callback()':
Marlin\src\HAL\STM32\HAL.cpp:170:3: error: 'hUSBHost' was not declared in this scope
  170 |   hUSBHost.Timer++;
      |   ^~~~~~~~
*** [.pio\build\tronxy_stm32f103\src\src\HAL\STM32\HAL.cpp.o] Error 1
```

Also, if compiled with STM32F446 no `update.cbd` is present and firmware is not flashed via SD card. 

### Version of Marlin Firmware

2?

borland1, 2022-11-22:
Same here. Also have XY2-Pro Titan and V6 (103 mainboard) with bootloader V1.56 and Tronxy firmware V1.24.

Please include support for how to modify file platformIO.ini to successfully compile and produce update folder for use with SD-card on the above mentioned board.

tronxy3d, 2022-11-26:
> ### Bug Description
> I have a XY2-PRO with Titan extruder, compiling the firmware with the STM32F446 board as target does not update the printer, I believe I have a 103 board (2 capacitators - chitu [...]

We have uploaded the source code of F103, please download it at the following website：
https://github.com/tronxy3d/STM32F103-PIC480x320

tronxy3d, 2022-11-26:
> 

> Please include support for `STM32F103` mainboard
> 
> Same as problem as `luigimannoni`.
> 
> Also, if I change extruder for `XY2-PRO` in file` TronxyMachine.h` to Titan, there are compile errors..
> 
> `#define USE_EXTRUDER 1 // Titan`
> 
> I also have XY2-Pro Titan and V6 (103 mainboard) with bootloader V1.56 and Tronxy firmware V1.24.
> 
> Please also include how to modify file platformIO.ini to successfully compile and produce update folder for use with SD-card on the above mentioned board.

I have uploaded the source code of F103, please download it at the following website：
https://github.com/tronxy3d/STM32F103-PIC480x320