tronxy3d/F4xx-SIM480x320 #2: XY 3 SE build FW flashing issue/semi-brick printer

MrKalach, 2022-11-24:
### Did you test the latest `bugfix-2.0.x` code?

Yes, and the problem still exists.

### Bug Description

Steps I did:
1) Clone the repo
2) Change `TRONXY_PROJ` to `PROJ_XY3SE_2E`
3) Build
4) Copy FW to SD card
5) Flash it

The problem was that at step 5, the printer didn't reboot automatically and stuck on 100% of flashing. After reboot, I see a grey screen with `Initialisation...` on the bottom.

My thoughts next, the original firmware has the next files:
```
fmw_xy3se.bin
font_tronxy.bin
log_tronxy.ypc
sd_guide.bin
ui_tronxy.ypc
```
Where `sd_guide.bin` has important information for the bootloader, where to place files in the flash. I also tried to replace `fmw_xy3se.bin` with compiled one, but the result was the same.

I think the repo still missing tools, to generate these legacy files.

Bootloader version: 1.60

### Version of Marlin Firmware

2.x

borland1, 2022-11-24:
You might need to request STM32F103 support.   Does you printer's mainboard have a `STM32F103` processor?

file README.md states....

`MCU: STM32F446 or GD32F4xx`

`STM32F103 is not supported`

MrKalach, 2022-11-25:
@borland1 I double-checked, and I have `466` SoC installed. 
Info screen from the printer (stock FW):
[image]

MrKalach, 2022-11-25:
Update:
I was able to make classic Marlin works!
1) Replace `TRONXY_UI` -> `UI_MARLIN_DEFAULT`
2) Do not comment out the next code:
```
void EXTI1_IRQHandler(void)
{
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_1);
}
```
Didn't do any test prints yet.

tronxy3d, 2022-11-26:
> 

First, make sure that your motherboard MCU is STM32F4 or GD32F4. Then, download the latest version of 2.18 and retry it. I uploaded the latest version of 2.18. If there is any abnormality, tell me. I'm sorry that I didn't return your information in time because I was too busy

tronxy3d, 2022-11-26:
> ### Bug Description
> Steps I did:
>
> 1. Clone the repo
> 2. Change `TRONXY_PROJ` to `PROJ_XY3SE_2E`
> 3. Build
> 4. Copy FW to SD card
> 5. Flash it
>
> The problem was that at step 5, the printer didn't reboot [...]

First, make sure that your motherboard MCU is STM32F4 or GD32F4. Then, download the latest version of 2.18 and retry it. I uploaded the latest version of 2.18. If there is any abnormality, tell me. I'm sorry that I didn't return your information in time because I was too busy

tronxy3d, 2022-11-26:
> You might need to request STM32F103 support. Does you printer's mainboard have a `STM32F103` processor?
> 
> file README.md states....
> 
> `MCU: STM32F446 or GD32F4xx`
> 
> `STM32F103 is not supported`

I update the source code which support stm32f103 just now and you can download the source code at the address:
https://github.com/tronxy3d/STM32F103-PIC480x320