tronxy3d/F4xx-SIM480x320 #3: Fehler beim erstellen des Builds

Riggy0815, 2022-11-27:
Hallo,

ich bekomme in VSCODE folgende fehlermeldung!

Archiving .pio\build\tronxy_stm32f446\lib6c8\libIWatchdog.a
Compiling .pio\build\tronxy_stm32f446\lib554\Wire\Wire.cpp.o
Archiving .pio\build\tronxy_stm32f446\lib5d4\libSPI.a
Compiling .pio\build\tronxy_stm32f446\lib554\Wire\utility\twi.c.o
Archiving .pio\build\tronxy_stm32f446\libade\libEEPROM.a
Archiving .pio\build\tronxy_stm32f446\libf85\libServo.a
Archiving .pio\build\tronxy_stm32f446\lib064\libSoftwareSerial.a
Archiving .pio\build\tronxy_stm32f446\lib554\libWire.a
Linking .pio\build\tronxy_stm32f446\firmware.elf
c:/users/richa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lchenyun
c:/users/richa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lmuyu
c:/users/richa/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lmcu
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1
================================================================== [FAILED] Took 109.05 seconds ==================================================================
Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:01:49.048
============================================================= 1 failed, 0 succeeded in 00:01:49.048 =============================================================

Aishor, 2022-11-27:
Same error
Archiving .pio\build\tronxy_stm32f446\lib35b\libSoftwareSerial.a
Archiving .pio\build\tronxy_stm32f446\lib3d6\libWire.a
Linking .pio\build\tronxy_stm32f446\firmware.elf
d:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lchenyun
d:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lmuyu
d:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot find -lmcu
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1

cembaykara, 2022-11-27:
yup

tronxy3d, 2022-11-28:
> Hallo,
>
> ich bekomme in VSCODE folgende fehlermeldung!
>
> Archiving .pio\build\tronxy_stm32f446\lib6c8\libIWatchdog.a Compiling .pio\build\tronxy_stm32f446\lib554\Wire\Wire.cpp.o Archiving [...]

Some uploaded files are missing, causing compilation exceptions. I'm sorry. I've uploaded them again. Please download again and compile again. It should be OK

tronxy3d, 2022-11-28:
> yup

Some uploaded files are missing, causing compilation exceptions. I'm sorry. I've uploaded them again. Please download again and compile again. It should be OK

cembaykara, 2022-11-28:
@tronxy3d Thanks. Got it to compile.

Riggy0815, 2022-11-28:
Besten Dank! Soweit funktioniert es!

BahamutWolf, 2022-12-25:
Ich erhalte beim Compilieren immer den Fehler:
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1

tronxy3d, 2022-12-26:
@BahamutWolf please see the FAQ

BahamutWolf, 2022-12-26:
@tronxy3d Danke es hat Funktioniert mit den FAQ`s.