tronxy3d/F4xx-SIM480x320 #6: Use UI_MARLIN_DEFAULT

wolfgangspiegel, 2022-11-30:
Hi, i have a X5SA_Pro. i try the default ui setting. The printer come up but restart every 10 sec. 
With the UI_SIM480X320 setting i had no problem, only if i whant to change the language. 
What could be!

tronxy3d, 2022-12-01:
> Hi, i have a X5SA_Pro. i try the default ui setting. The printer come up but restart every 10 sec. With the UI_SIM480X320 setting i had no problem, only if i whant to change the language. What could be!

But I use the machine to test it. It's OK.
[update.zip](https://github.com/tronxy3d/F4xx-SIM480x320/files/10129039/update.zip)

wolfgangspiegel, 2022-12-01:
Hi. ThanksI try your build. Sorry its better but not stable. The org firmware and the build with UI_SIM480X320 is stable. 
But Thanks i wait for the next relaese.

wolfgangspiegel, 2022-12-26:
It must have something to do with the mess with the board types!! I have two mainboards of the type F446Z both have the stability problem makes no difference which UI is used.
Just typical Tronxy

Infobitas, 2023-02-10:
Hi. 
I also have problems while using UI_MARLIN_DEFAULT.
Printer UI loads ok after firmware flash, but screen goes white (shows nothing) after power off and on. Switching to UI_SIM480X320 fixes problem and screen is working again. 

PROJ_X5SA_PRO with BOARD_TRONXY_F446

tronxy3d, 2023-03-09:
We have made some changes to the firmware, if you like marlin default UI, we recommend you use the official marlin, and then refer to our open source code to search the UI_ MARLIN_ DEFAULT related control code block, change LCD configuration and pin configuration to the official firmware, then you can run the official firmware on our machine.
NOTE: you must set board_build.offset to 0x10000 in platformio.ini, and compile it to hex file, take the hex file to the update directory