tronxy3d/F4xx-SIM480x320 #9: multiple definition of `EXTI1_IRQHandler

reinhold1996, 2022-12-15:
Hello :)

Ive got errors on 2 different Computers. I´d like to have the standard marlin ui and some tweaks like UBL. So i´d try to compile and got those error:

Linking .pio\build\tronxy_stm32f446\firmware.elf
c:/users/reinhold/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/muyu\libmuyu.a(muyu.cpp.o): in function `EXTI1_IRQHandler':
F:\YSZ\Work\3Firmware-2.1.1/Marlin\muyu/muyu.cpp:212: multiple definition of `EXTI1_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o:C:\Users\Reinhold\.platformio\packages\framework-arduinoststm32@src-37335c727965154be67d1d5ed653b9c6\libraries\SrcWrapper\src\stm32/interrupt.cpp:339: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1

i´d attach a .txt of the full history. And some imgs from my tronxy board/system menue. Pls help me :(

[console_log.txt](https://github.com/tronxy3d/F4xx-SIM480x320/files/10236006/console_log.txt)
[image]
[image]
[image]

closed 2022-12-15 by reinhold1996

reinhold1996, 2022-12-15:
pls delete i´d overlook

tronxy3d, 2022-12-20:
@reinhold1996 Please refer to the FAQ to resolve the problem