tronxy3d/F4xx-SIM480x320 #12: AUTO_BED_LEVELING_UBL on version 2.18

lulek2007, 2022-12-28:
Dear Sirs,

I have a Tronxy X5SA-400 Pro printer.
I am very glad that the Marlin software has been made available here

I downloaded and compiled the November version (SIM480x320_2.17.3) https://github.com/tronxy3d/F4xx-SIM480x320 for my printer and enable MARLIN_UI
In addition to leveling the bed, I modified the Configuration.h file and turned on AUTO_BED_LEVELING_UBL.

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#if TRONXY_PROJ != PROJ_XY3_PRO \
&& TRONXY_PROJ != PROJ_D01 \
&& TRONXY_PROJ != PROJ_C2 \
&& TRONXY_PROJ != PROJ_MOORE1 \
&& TRONXY_PROJ != PROJ_X5SA_g \
&& TRONXY_PROJ != PROJ_MOORE2_PRO \
&& TRONXY_PROJ != PROJ_TAONI
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
#endif

This is a very useful option in large printers. I have it in Creality CR10-s5 and after enabling this option, bed leveling is precise when printing.

I wanted to do the same in December version (SIM480x320_2.18.0)

https://github.com/tronxy3d/F4xx-SIM480x320/releases/tag/SIM480x320-2.18.0r

in the Configuration.h file I change

//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
#if ENABLED(FIX_MOUNTED_PROBE)
//#define MESH_BED_LEVELING
//#define AUTO_BED_LEVELING_BILINEAR
#define AUTO_BED_LEVELING_UBL
#endif

Unfortunately when compiling with UBL I get

Compiling .pio/build/tronxy_stm32f446/src/src/module/temperature.cpp.o
Marlin/src/module/settings.cpp: In static member function 'static void MarlinSettings::store_mesh(int8_t)':
Marlin/src/module/settings.cpp:2888:87: error: no matching function for call to 'PersistentStore::write_data(int&, uint8_t* const&, unsigned int, uint16_t*)'
  2888 | const bool status = persistentStore.write_data(pos, src, MESH_STORE_SIZE, &crc);
       | ^
In file included from Marlin/src/module/settings.h:31,
                  from Marlin/src/module/settings.cpp:45:
Marlin/src/module/../HAL/shared/eeprom_api.h:43:15: note: candidate: 'static bool PersistentStore::write_data(int&, const uint8_t*, size_t)'
    43 | static bool write_data(int &pos, const uint8_t *value, size_t size);
       | ^~~~~~~~~~
Marlin/src/module/../HAL/shared/eeprom_api.h:43:15: note: candidate expects 3 arguments, 4 provided
Marlin/src/module/../HAL/shared/eeprom_api.h:68:22: note: candidate: 'static bool PersistentStore::write_data(int, uint8_t)'
    68 | static inline bool write_data(const int pos, const uint8_t value) {
       | ^~~~~~~~~~
Marlin/src/module/../HAL/shared/eeprom_api.h:68:22: note: candidate expects 2 arguments, 4 provided
Marlin/src/module/settings.cpp: In static member function 'static void MarlinSettings::load_mesh(int8_t, void*)':
Marlin/src/module/settings.cpp:2922:81: warning: the address of 'crc' will never be NULL [-Waddress]
  2922 | uint16_t status = persistentStore.read_data(pos, dest, MESH_STORE_SIZE, &crc);
       | ^~~~
*** [.pio/build/tronxy_stm32f446/src/src/module/settings.cpp.o] Error 1
===================================== [FAILED] Took 91.80 seconds ==== ==================================

Environment Status Duration
---------------- -------- ------------
tronxy_stm32f446 FAILED 00:01:31.799

Can I please correct the code so that it compiles correctly with the AUTO_BED_LEVELING_UBL option

Regards, Marcin Kowalik

tronxy3d, 2022-12-29:
Please download the firmware again for compilation

lulek2007, 2022-12-29:
Thank you very much,

Now the firmware is compiling

The only thing I noticed is missing the z_offset icon in tools (disabled in Configuration_adv.h). For me it's not a problem because I set z_offset in Octoprint init script.

Regards MK

lulek2007, 2022-12-29:
Hi,

I found a bug in the patched software

G29 command does not work correct , resets the printer.

I use standard commands for UBL leveling

G28 ; home all axes
M420 S0 ; Turning off bed leveling while probing, if firmware is set
           ; to restore after G28
M155 S30 ; reduce temperature reporting rate to reduce output pollution
M190 S65 ; (optional) wait for the bed to get up to temperature
G29 P1 ; automatically populate mesh with all reachable points
G29 P3 ; infer the rest of the mesh values
G29 P3 ; infer the rest of the mesh values again
@ BEDLEVELVISUALIZER ; tell the plugin to watch for reported mesh
M420 S1 V ; enabled leveling and report the new mesh
G29 S0 ; Save UBL mesh points to slot 0 (EEPROM).
G29 F 10.0 ; Set Fade Height for correction at 10.0 mm.
G29A; Activate the UBL System.
M500 ; save the current setup to EEPROM
M155 S3; reset temperature reporting
M140 S0 ; cooling down the bed

the printer resets on the command

G29 S0 ; Save UBL mesh points to slot 0 (EEPROM).

the same happens when reading G29 L0

weird but command G29 S work (from marlin manual - If no slot number is given, save to the last-activated slot)

Regards
MK

nemesis567, 2023-02-03:
I also have this issue! But g29 S saves nothing and g29 s4 also reboots it.