tronxy3d/F4xx-SIM480x320 #13: filament sensor does not work when printing with octoprint

lulek2007, 2023-01-09:
Hi, 

I have another problem when printing with firmware 2.18
When I print from the SD card and the filament runs out, the sensor works and the printer asks to replace the filament. When I print with octoprint and the filament runs out, the printer does not stop printing. Can I ask for help on how to enable this functionality in the shared software?

In the terminal after issuing the m119 command and the filament inserted I have
Send: M119
Recv: Reporting endstop status
recv: x_min: TRIGGERED
Recv: y_min: TRIGGERED
recv: z_min: open
Recv: filament: TRIGGERED
recv: ok

or without filament

Send: M119
Recv: Reporting endstop status
recv: x_min: TRIGGERED
Recv: y_min: TRIGGERED
recv: z_min: open
recv: filament: open
recv: ok

so the sensor is working

I also check the M412

recv: echo:Filament runout ON ; Host Handling ON
recv: ok

I do not understand why the printer does not stop printing when there is no filament and printing with octoprint

Regards
MK

lulek2007, 2023-02-08:
Maybe try G29 S0 or S1 or S2 etc.
https://marlinfw.org/docs/gcode/G029-ubl.html

I uploaded older firmware 2.0.9. 2.17.3 https://github.com/tronxy3d/F4xx-SIM480x320/releases/tag/release

and it works with UBL ok

tronxy3d, 2023-03-09:
make sure the FILAMENT_RUNOUT_CHECK be 1 in platformio.ini, and the FIX_MOUNTED_PROBE is defined in TronxyMachine.h