tronxy3d/F4xx-SIM480x320 #15: Compile Issue

Alphacode, 2023-03-08:
after updating TronxyMachine.h it does not update the build, i have to manually configure Configuration.h with my board, i have to also manually configure machine in Configuration.h, there is no auto updating being done after editing TronxyMachine.h

tronxy3d, 2023-03-09:
Please refer to the description of the webpage and follow the steps

Alphacode, 2023-03-10:
I have followed all instructions and the compile completes with an error but still completes but the marlin autobuild shows a different board and printer, I have an X5SA_Pro with the TronXY CXYV6-191121 board. the error after build completes is  see pic.
[image]