tronxy3d/F4xx-SIM480x320 #16: Issue with compiling

condescomputers, 2023-04-02:
I run win 10, with a Tronxy X5sa pro with the F446 chip.  I followed the directions aboutComenting out lines for the IRQ issue, and after I do this"

// void EXTI1_IRQHandler(void)
// {
//  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_1);
// }

I get this error" 

Linking .pio\build\tronxy_stm32f446\firmware.elf
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `std::function<void ()>::operator bool() const':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:221: multiple definition of `stm32_interrupt_disable(GPIO_TypeDef*, 
unsigned short)'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:221: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `HAL_GPIO_EXTI_Callback':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:241: multiple definition of `HAL_GPIO_EXTI_Callback'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy 
- Copy.cpp:241: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI0_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:329: multiple definition of `EXTI0_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:329: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI2_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:349: multiple definition of `EXTI2_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:349: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI3_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:359: multiple definition of `EXTI3_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:359: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI4_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:369: multiple definition of `EXTI4_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:369: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI9_5_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:379: multiple definition of `EXTI9_5_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:379: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `EXTI15_10_IRQHandler':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:392: multiple definition of `EXTI15_10_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - 
Copy.cpp:392: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `stm32_interrupt_enable(GPIO_TypeDef*, unsigned short, std::function<void ()>, unsigned long)':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:129: multiple definition of `stm32_interrupt_enable(GPIO_TypeDef*, unsigned short, std::function<void ()>, unsigned long)'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:129: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o: in function `stm32_interrupt_enable(GPIO_TypeDef*, unsigned short, void (*)(), unsigned long)':
C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:208: multiple definition of `stm32_interrupt_enable(GPIO_TypeDef*, unsigned short, void (*)(), unsigned long)'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:208: first defined here
c:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/muyu\libmuyu.a(muyu.cpp.o): in function `EXTI1_IRQHandler':
F:\YSZ\Work\3Firmware-2.1.1/Marlin\muyu/muyu.cpp:212: multiple definition of `EXTI1_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt - Copy - Copy.cpp.o:C:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt - Copy - Copy.cpp:339: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1
================================================================== [FAILED] Took 274.14 seconds ==================================================================
Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:04:34.137
============================================================= 1 failed, 0 succeeded in 00:04:34.137 ============================================================= 

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
"

All ive done is follow the directions.

Just in case above didnt paste engough:

#ifdef __cplusplus
extern "C" {
#endif
/**
  * @brief This function handles external line 0 interrupt request.
  * @param  None
  * @retval None
  */
void EXTI0_IRQHandler(void)
{
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_0);
}

/**
  * @brief This function handles external line 1 interrupt request.
  * @param  None
  * @retval None
  */
// void EXTI1_IRQHandler(void)
// {
//  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_1);
// }

/**
  * @brief This function handles external line 2 interrupt request.
  * @param  None
  * @retval None
  */
void EXTI2_IRQHandler(void)
{
  HAL_GPIO_EXTI_IRQHandler(GPIO_PIN_2);
}

condescomputers, 2023-04-02:
I made a mistake on that first compile.  but now I get this:

6\src\src\gcode\temp\M155.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\gcode\temp\M303.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\gcode\units\M82_M83.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\inc\Warnings.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\lcd\extui\ui_api.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\lcd\fontutils.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\lcd\marlinui.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\buzzer.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\crc16.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\nozzle.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\numtostr.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\stopwatch.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\libs\vector_3.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\endstops.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\motion.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\planner.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\probe.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\settings.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\stepper.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\stepper\indirection.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\temperature.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\module\tool_change.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\Sd2Card.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdBaseFile.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdFatUtil.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdFile.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdVolume.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\cardreader.cpp.o
Compiling .pio\build\tronxy_stm32f446\src\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp.o
Compiling .pio\build\tronxy_stm32f446\lib774\IWatchdog\IWatchdog.cpp.o
Compiling .pio\build\tronxy_stm32f446\lib11d\SPI\SPI.cpp.o
Archiving .pio\build\tronxy_stm32f446\lib774\libIWatchdog.a
Compiling .pio\build\tronxy_stm32f446\lib11d\SPI\utility\spi_com.c.o
Compiling .pio\build\tronxy_stm32f446\lib66b\EEPROM\utility\stm32_eeprom.c.o
Compiling .pio\build\tronxy_stm32f446\liba7d\SoftwareSerial\SoftwareSerial.cpp.o
Archiving .pio\build\tronxy_stm32f446\lib11d\libSPI.a
Compiling .pio\build\tronxy_stm32f446\libb9e\Servo\stm32\Servo.cpp.o
Compiling .pio\build\tronxy_stm32f446\lib62f\Wire\Wire.cpp.o
Archiving .pio\build\tronxy_stm32f446\lib66b\libEEPROM.a
Compiling .pio\build\tronxy_stm32f446\lib62f\Wire\utility\twi.c.o
Archiving .pio\build\tronxy_stm32f446\liba7d\libSoftwareSerial.a
Archiving .pio\build\tronxy_stm32f446\libb9e\libServo.a
Archiving .pio\build\tronxy_stm32f446\lib62f\libWire.a
Linking .pio\build\tronxy_stm32f446\firmware.elf
Building update/fmw_tronxy.bin
Usage: arm-none-eabi-objcopy [option(s)] in-file [out-file]
 Copies a binary file, possibly transforming it in the process
 The options are:
  -I --input-target <bfdname>      Assume input file is in format <bfdname>
  -O --output-target <bfdname>     Create an output file in format <bfdname>
  -B --binary-architecture <arch>  Set output arch, when input is arch-less
  -F --target <bfdname>            Set both input and output format to <bfdname>
     --debugging                   Convert debugging information, if possible
  -p --preserve-dates              Copy modified/access timestamps to the output
  -D --enable-deterministic-archives
                                   Produce deterministic output when stripping archives
  -U --disable-deterministic-archives
                                   Disable -D behavior (default)
  -j --only-section <name>         Only copy section <name> into the output
     --add-gnu-debuglink=<file>    Add section .gnu_debuglink linking to <file>
  -R --remove-section <name>       Remove section <name> from the output
     --remove-relocations <name>   Remove relocations from section <name>
  -S --strip-all                   Remove all symbol and relocation information
  -g --strip-debug                 Remove all debugging symbols & sections
     --strip-dwo                   Remove all DWO sections
     --strip-unneeded              Remove all symbols not needed by relocations
  -N --strip-symbol <name>         Do not copy symbol <name>
     --strip-unneeded-symbol <name>
[... 81 lines cut ...]
     --subsystem <name>[:<version>]
                                   Set PE subsystem to <name> [& <version>]
     --compress-debug-sections[={none|zlib|zlib-gnu|zlib-gabi}]
                                   Compress DWARF debug sections using zlib
     --decompress-debug-sections   Decompress DWARF debug sections using zlib
     --elf-stt-common=[yes|no]     Generate ELF common symbols with STT_COMMON
                                     type
     --verilog-data-width <number> Specifies data width, in bytes, for verilog output
  -M  --merge-notes                Remove redundant entries in note sections
      --no-merge-notes             Do not attempt to remove redundant notes (default)
  -v --verbose                     List all object files modified
  @<file>                          Read options from <file>
  -V --version                     Display this program's version number
  -h --help                        Display this output
[... cut ...]
================================================================== [FAILED] Took 262.62 seconds ==================================================================
Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:04:22.623
============================================================= 1 failed, 0 succeeded in 00:04:22.623 ============================================================= 

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

bytefix, 2023-04-24:
Hi @condescomputers try this: [https://github.com/tronxy3d/F4xx-SIM480x320/issues/17#issuecomment-1520708237](url)

Maker-Paul, 2024-11-22:
> I made a mistake on that first compile. but now I get this:
> 
> 6\src\src\gcode\temp\M155.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\gcode\temp\M303.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\gcode\units\M82_M83.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\inc\Warnings.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\lcd\extui\ui_api.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\lcd\fontutils.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\lcd\marlinui.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\buzzer.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\crc16.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\nozzle.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\numtostr.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\stopwatch.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\libs\vector_3.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\endstops.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\motion.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\planner.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\probe.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\settings.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\stepper.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\stepper\indirection.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\temperature.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\module\tool_change.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\Sd2Card.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdBaseFile.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdFatUtil.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdFile.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\SdVolume.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\cardreader.cpp.o Compiling .pio\build\tronxy_stm32f446\src\src\sd\usb_flashdrive\Sd2Card_FlashDrive.cpp.o Compiling .pio\build\tronxy_stm32f446\lib774\IWatchdog\IWatchdog.cpp.o Compiling .pio\build\tronxy_stm32f446\lib11d\SPI\SPI.cpp.o Archiving .pio\build\tronxy_stm32f446\lib774\libIWatchdog.a Compiling .pio\build\tronxy_stm32f446\lib11d\SPI\utility\spi_com.c.o Compiling .pio\build\tronxy_stm32f446\lib66b\EEPROM\utility\stm32_eeprom.c.o Compiling .pio\build\tronxy_stm32f446\liba7d\SoftwareSerial\SoftwareSerial.cpp.o Archiving .pio\build\tronxy_stm32f446\lib11d\libSPI.a Compiling .pio\build\tronxy_stm32f446\libb9e\Servo\stm32\Servo.cpp.o Compiling .pio\build\tronxy_stm32f446\lib62f\Wire\Wire.cpp.o Archiving .pio\build\tronxy_stm32f446\lib66b\libEEPROM.a Compiling .pio\build\tronxy_stm32f446\lib62f\Wire\utility\twi.c.o Archiving .pio\build\tronxy_stm32f446\liba7d\libSoftwareSerial.a Archiving .pio\build\tronxy_stm32f446\libb9e\libServo.a Archiving .pio\build\tronxy_stm32f446\lib62f\libWire.a Linking .pio\build\tronxy_stm32f446\firmware.elf Building update/fmw_tronxy.bin Usage: arm-none-eabi-objcopy [option(s)] in-file [out-file] Copies a binary file, possibly transforming it in the process The options are: -I --input-target Assume input file is in format -O --output-target Create an output file in format -B --binary-architecture Set output arch, when input is arch-less -F --target Set both input and output format to --debugging Convert debugging information, if possible -p --preserve-dates Copy modified/access timestamps to the output -D --enable-deterministic-archives Produce deterministic output when stripping archives -U --disable-deterministic-archives Disable -D behavior (default) -j --only-section Only copy section into the output --add-gnu-debuglink= Add section .gnu_debuglink linking to -R --remove-section Remove section from the output --remove-relocations Remove relocations from section -S --strip-all Remove all symbol and relocation information -g --strip-debug Remove all debugging symbols & sections --strip-dwo Remove all DWO sections --strip-unneeded Remove all symbols not needed by relocations -N --strip-symbol Do not copy symbol --strip-unneeded-symbol Do not copy symbol unless needed by relocations --only-keep-debug Strip everything but the debug information --extract-dwo Copy only DWO sections --extract-symbol Remove section contents but keep symbols -K --keep-symbol Do not strip symbol --keep-file-symbols Do not strip file symbol(s) --localize-hidden Turn all ELF hidden symbols into locals -L --localize-symbol Force symbol to be marked as a local --globalize-symbol Force symbol to be marked as a global -G --keep-global-symbol Localize all symbols except -W --weaken-symbol Force symbol to be marked as a weak --weaken Force all global symbols to be marked as weak -w --wildcard Permit wildcard in symbol comparison -x --discard-all Remove all non-global symbols -X --discard-locals Remove any compiler-generated symbols -i --interleave[=] Only copy N out of every bytes --interleave-width Set N for --interleave -b --byte Select byte in every interleaved block --gap-fill Fill gaps between sections with --pad-to Pad the last section up to address --set-start Set the start address to {--change-start|--adjust-start} Add to the start address {--change-addresses|--adjust-vma} Add to LMA, VMA and start addresses {--change-section-address|--adjust-section-vma} {=|+|-} Change LMA and VMA of section by --change-section-lma {=|+|-} Change the LMA of section by --change-section-vma {=|+|-} Change the VMA of section by {--[no-]change-warnings|--[no-]ad [... cut ...]
> ================================================================== [FAILED] Took 262.62 seconds ==================================================================
> Environment Status Duration
> 
> tronxy_stm32f446 FAILED 00:04:22.623 ============================================================= 1 failed, 0 succeeded in 00:04:22.623 =============================================================
> 
>     * The terminal process "C:.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1.
> 
>     * Terminal will be reused by tasks, press any key to close it.

The suggested fix in #17(comment) has been corrected and update