tronxy3d/F4xx-SIM480x320 #19: vscode on linux : compile issues

flatmax, 2023-05-15:
Is it possible I am missing some python requirements ?

Processing tronxy_stm32f446 (board: tronxy_stm32f446zet; platform: ststm32@~12.1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/tronxy_stm32f446zet.html
PLATFORM: ST STM32 (12.1.1) > STM32F446ZE (128k RAM. 512k Flash)
HARDWARE: STM32F446ZET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32 @ 0.0.0 
 - framework-cmsis @ 2.50501.200527 (5.5.1) 
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "/home/flatmax/.platformio/penv/lib/python3.10/site-packages/platformio/builder/main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "/home/flatmax/temp/F4xx-SIM480x320/D:/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/flatmax/temp/F4xx-SIM480x320/D:/.platformio/packages/tool-scons/scons-local-4.5.2/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/flatmax/temp/F4xx-SIM480x320/buildroot/share/PlatformIO/scripts/tronxy_scrypt.py", line 9:
    marlin.relocate_vtab(board.get('build.offset'))
  File "/home/flatmax/temp/F4xx-SIM480x320/buildroot/share/PlatformIO/scripts/marlin.py", line 30:
    replace_define("VECT_TAB_OFFSET", address)
  File "/home/flatmax/temp/F4xx-SIM480x320/buildroot/share/PlatformIO/scripts/marlin.py", line 19:
    for define in env['CPPDEFINES']:
=============================================================== [FAILED] Took 1.57 seconds ===============================================================

Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:00:01.569
========================================================= 1 failed, 0 succeeded in 00:00:01.569 =========================================================

 *  The terminal process "platformio 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

Blackcatswiskas, 2023-05-23:
Have you managed to find an answer yet? I think we have the same issue.

flatmax, 2023-05-23:
nope!

cyclemat, 2023-08-24:
same issue here !

g4m3r0, 2024-08-11:
@cyclemat this one fixed it for me https://github.com/tronxy3d/F4xx-SIM480x320/issues/20#issuecomment-1567568743