tronxy3d/F4xx-SIM480x320 #20: Cant get it to compile/build

Blackcatswiskas, 2023-05-19:
I am having real issues here I received a compiled bin file from someone for the X5sa 500pro with 466 bit it only turns on z motor so tried to compile myself only to get errors what am i doing wrong?
Error: Traceback (most recent call last):
  File "C:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 102, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "C:\.platformio\penv\lib\site-packages\platformio\cli.py", line 70, in invoke
    return super().invoke(ctx)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1654, in invoke
    super().invoke(ctx)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\.platformio\penv\lib\site-packages\click\core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\click\decorators.py", line 26, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\__main__.py", line 56, in cli
    maintenance.on_platformio_start(ctx, caller)
  File "C:\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 37, in on_platformio_start   
    telemetry.on_command()
  File "C:\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 275, in on_command
    resend_backuped_reports()
  File "C:\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 427, in resend_backuped_reports
    tm = app.get_state_item("telemetry", {})
  File "C:\.platformio\penv\lib\site-packages\platformio\app.py", line 181, in get_state_item
    with State() as state:
  File "C:\.platformio\penv\lib\site-packages\platformio\app.py", line 81, in __init__
    os.makedirs(core_dir)
  File "C:\.platformio\python3\lib\os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "C:\.platformio\python3\lib\os.py", line 225, in makedirs
    mkdir(name, mode)
PermissionError: [WinError 21] The device is not ready: 'D:\\'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "C:\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 390, in _finalize
    if not MPDataPusher().in_wait():
  File "C:\.platformio\penv\lib\site-packages\platformio\util.py", line 132, in get_instance
    _instances[cls] = cls(*args, **kwargs)
  File "C:\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 196, in __init__
    self._http_session = HTTPSession()
  File "C:\.platformio\penv\lib\site-packages\platformio\http.py", line 53, in __init__
    self.verify = app.get_setting("enable_proxy_strict_ssl")
  File "C:\.platformio\penv\lib\site-packages\platformio\app.py", line 202, in get_setting
    with State() as state:
  File "C:\.platformio\penv\lib\site-packages\platformio\app.py", line 81, in __init__
    os.makedirs(core_dir)
  File "C:\.platformio\python3\lib\os.py", line 215, in makedirs
    makedirs(head, exist_ok=exist_ok)
  File "C:\.platformio\python3\lib\os.py", line 225, in makedirs
    mkdir(name, mode)
PermissionError: [WinError 21] The device is not ready: 'D:\\'

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

Blackcatswiskas, 2023-05-21:
latest errors
 File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\tinas\Downloads\F4xx-SIM480x320-SIM480x320-2.18.0r\F4xx-SIM480x320-SIM480x320-2.18.0r\buildroot\share\PlatformIO\scripts\tronxy_scrypt.py", line 8:
    marlin.relocate_vtab(board.get('build.offset'))
  File "C:\Users\tinas\Downloads\F4xx-SIM480x320-SIM480x320-2.18.0r\F4xx-SIM480x320-SIM480x320-2.18.0r\buildroot\share\PlatformIO\scripts\marlin.py", line 30:
    replace_define("VECT_TAB_OFFSET", address)
  File "C:\Users\tinas\Downloads\F4xx-SIM480x320-SIM480x320-2.18.0r\F4xx-SIM480x320-SIM480x320-2.18.0r\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:

EQNish, 2023-05-25:
Same, just tried to compile and get the exact same errors
`FIGURATION: https://docs.platformio.org/page/boards/ststm32/tronxy_stm32f446zet.html
PLATFORM: ST STM32 (12.1.1) > STM32F446ZE (128k RAM. 512k Flash)
HARDWARE: STM32F446ZET6 180MHz, 128KB RAM, 512KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES:
 - framework-arduinoststm32 @ 0.0.0
 - framework-cmsis @ 2.50501.200527 (5.5.1)
 - toolchain-gccarmnoneeabi @ 1.90201.191206 (9.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 10 compatible libraries
Scanning dependencies...
Dependency Graph
|-- IWatchdog @ 1.0.0
|-- SPI @ 1.0
|-- EEPROM @ 2.0.1
|-- SoftwareSerial @ 1.0
|-- Servo @ 1.1.2
|-- Wire @ 1.0
Building in release mode
RuntimeError: deque mutated during iteration:
  File "C:\Users\eqnish\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 181:
    env.SConscript(env.GetExtraScripts("post"), exports="env")
  File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "D:\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\eqnish\OneDrive\3dPrinting\Printers\TRONXY_X5SA_330-main\F4xx-SIM480x320-firmware\buildroot\share\PlatformIO\scripts\tronxy_scrypt.py", line 9:
    marlin.relocate_vtab(board.get('build.offset'))
  File "C:\Users\eqnish\OneDrive\3dPrinting\Printers\TRONXY_X5SA_330-main\F4xx-SIM480x320-firmware\buildroot\share\PlatformIO\scripts\marlin.py", line 30:
    replace_define("VECT_TAB_OFFSET", address)
  File "C:\Users\eqnish\OneDrive\3dPrinting\Printers\TRONXY_X5SA_330-main\F4xx-SIM480x320-firmware\buildroot\share\PlatformIO\scripts\marlin.py", line 19:
    for define in env['CPPDEFINES']:
==================================================== [FAILED] Took 5.87 seconds ====================================================

Environment       Status    Duration`

Tukamok, 2023-05-29:
I'm getting the same.  I have had prior success with getting this to build, but something has changed recently and I'm not sure exactly what.

Tukamok, 2023-05-29:
The issue appears to be related to an update to PlatformIO.  Since it autoupdates rolling back might be difficult, but there appears to be a fix [here](https://community.platformio.org/t/trying-to-compile-marlin-2-1-2-for-ender-3-skr-mini-v2-0-board/33769/8)

specifically maxgerhardt's second response:

replace 

def replace_define in line 16 (or so) in buildroot/share/PlatformIO/scripts/marlin.py

with 

def replace_define(field, value):
    envdefs = env['CPPDEFINES'].copy()
    for define in envdefs:
        if define[0] == field:
            env['CPPDEFINES'].remove(define)
    env['CPPDEFINES'].append((field, value))