tronxy3d/F4xx-SIM480x320 #23: Got it compiling and working, some hints below:

SyncroSeb, 2023-07-31:
1.	Fresh install
Needed a fresh install of VS Code with PlatformIO IDE on a different clean windows laptop, wasn’t working on my normal work machine

2.	Marlin.py
Replaced Tronxys Version with Original from
https://github.com/MarlinFirmware/Marlin/blob/c2decc3e2e30c7cb0f517b7e40d8138a8c1d4a81/buildroot/share/PlatformIO/scripts/marlin.py

3.	Bootloop - platformio.ini
Line 313 modify -DUSBHOST_HS_EN=1 to -DUSBHOST_HS_EN=0

After that I got it compiling for my X5SA Pro without any other hickups

masterpj555, 2024-07-22:
Tried that with a fresh install of both vs code and platform io
Even went as far as removing reference to the tronxy script... it compiles for a bit but then

d:/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: Marlin/muyu\libmuyu.a(muyu.cpp.o): in function `EXTI1_IRQHandler':
F:\YSZ\Work\3Firmware-2.1.1/Marlin\muyu/muyu.cpp:212: multiple definition of `EXTI1_IRQHandler'; .pio/build/tronxy_stm32f446/SrcWrapper/src/stm32/interrupt.cpp.o:D:\.platformio\packages\framework-arduinoststm32\libraries\SrcWrapper\src\stm32/interrupt.cpp:339: first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\tronxy_stm32f446\firmware.elf] Error 1
============================================================ [FAILED] Took 56.55 seconds ============================================================

Environment       Status    Duration
----------------  --------  ------------
tronxy_stm32f446  FAILED    00:00:56.550

Tried for X5SA but doesn't matter which device i compile for

Maker-Paul, 2024-11-22:
> Tried that with a fresh install of both vs code and platform io Even went as far as removing reference to the tronxy script... it compiles for a bit but then [...]

You probably have a space character in your path to the project folder. That command is executed after compilation to adjust the output binary and copy it to the proper output folder. So it has compiled correctly It is a low level command that can't resolve path's that contain spaces. so it cant find the source file.