tronxy3d/F4xx-SIM480x320 #29: Uncoorrect Gcode comments interpritation

lselis, 2023-10-23:
Hello everyone!
It`s not a promblem, it`s annoying bug...
If start gcode look`s like
.......
.....
G28
....
.....
everything is OK. But if line with G28 get comment, after parking X and Y printhead don`t move to center, and Z parking happened in X=0 and =0 coordinate. Without X and Y offset sensor haven`t bed under yourself, and printer try to destroy yourself.....
Is it firmware trouble? Or hardware?

P.S. printer model X5SA-Ug, board CXY-446-v10-220413, firmware 2.18.0 as buying

mweimholt, 2025-01-05:
I had this issue as well. X5SA-Pro with CXY-V6-191121, STM32F446ZET, and firmware 2.18.0.

Just as @lselis said, if the G28 command is on a line by itself with no trailing comment, everything works fine. With a comment, the print head gets stuck in the corner, and when the bed moves up it doesn't trigger the Z-axis sensor, because the sensor is positioned beyond the edge of the bed. So the bed just keeps moving up and crashes into the print head.

I discovered this because the start code for the X5SA Pro in Cura 5.9 is commented (see attached image). Removing the comment allows the print head to move back to the center of the bed, thus allowing the Z sensor to detect the bed and prevent the collision.

[image]

navlisData, 2026-11-01:
I can confirm the same issue on a **Tronxy XY-2 Pro 2E** (XY2-Pro-2E-UT, Marlin 2.1.1 / Tronxy V2.18.0, build November 28, 2022), using G-code generated by Cura 5.13.0 and printed from an SD card.

### Reproduction

- `G28 ; Home all axis` homes X/Y, then starts Z homing in the front-left corner, outside the bed's probe detection area. I had to stop the printer to prevent a collision.
- Changing **only** that line to `G28` (removing the comment) fixes the problem: the print head moves over the bed before probing Z.

Both behaviors were reproduced on my printer.

### Suspected root cause

I traced the issue through the published firmware source:

1. [queue.cpp, lines 365–370](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/queue.cpp#L365-L370)](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/queue.cpp#L365-L370) Semicolon comment stripping is disabled when `TRONXY_UI` is enabled. The comment is passed to the G-code parser.
2. [parser.cpp, lines 295–363](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/parser.cpp#L295-L363)](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/parser.cpp#L295-L363) The parser consumes `;` using `*p++`, then skips whitespace, advancing to `H` in `Home`. The subsequent `if (*p == ';') break` check comes too late. On the next iteration, `H` is registered as a parameter.
3. [G28.cpp, lines 520–526](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/calibrate/G28.cpp#L520-L526)](https://github.com/tronxy3d/F4xx-SIM480x320/blob/f490699bfed17f4ae1ad0cb5895db5c77448434c/Marlin/src/gcode/calibrate/G28.cpp#L520-L526) `parser.seen_test('H')` then bypasses `home_z_safely()` and calls `homeaxis(Z_AXIS)` at the current XY position.

This explains why a harmless comment changes the homing behavior and can cause a physical collision.

### Suggested fix

- Strip semicolon comments before parsing, including when `TRONXY_UI` is enabled.
- Additionally, stop parameter parsing immediately when encountering `;`.

The code references are from the last publicly available 2023 firmware source. I cannot confirm whether newer, unpublished firmware versions have already fixed this issue.