tronxy3d/F4xx-SIM480x320 #30: platformio.ini line 15 D directory

VKHelldog, 2024-01-01:
I don't know why this is line 15 core_dir     = D:/.platformio   It made my stuff not work. I deleted it and saved and now it seems to be working.

SyncroSeb, 2024-01-14:
changed it to C:/  that got it starting

Maker-Paul, 2024-11-22:
this is the .platformio directory that is built by VScode for this project. that way you can have different content for different projects. You need to set the path to a drive you have. If you want to keep the directory rather than overwriting it you can just use ".platformio" without a path and it will build the directory in your project folder. It makes it much faster to load when you switch between projects because it doesn't have to rebuild the folder. It is however a large folder so can use up disk space. and you will need to delete it if you want to do a completely clean new build.